ardera/flutterpi_tool #1: Inability to Tree Shake Icons Fonts in Release Mode

TheKiHub, 2023-07-05:
Hey, application creation tool that functions well in debug mode. 
However, when attempting to build applications in release mode, an error occurs stating, "This application cannot tree shake icons fonts." The "--no-tree-shake-icons" flag needs to be added to resolve this error.

Unfortunately, the flutterpi-tool does not currently support the inclusion of the "--no-tree-shake-icons" flag. This limitation prevents users from successfully building Flutter applications in release mode, particularly when using the "json_theme" package, which relies on the flag (package link: https://pub.dev/packages/json_theme).

Adding support for the "--no-tree-shake-icons" flag within the flutterpi-tool would greatly enhance its functionality and allow users to create release builds without encountering the aforementioned error.

Please let me know if any additional information is required.

ardera, 2023-07-05:
Oh, I actually didn't know there was any real use for that flag, but that totally makes sense. I'll add it.

ardera, 2023-07-05:
- implemented: https://github.com/ardera/flutterpi_tool/commit/0a6f8ef0998ae615eeef8e2a2c7f53084c5a75c1

Run `flutter pub global activate flutterpi_tool` again to get the latest version.

closed 2023-07-05 by ardera