ardera/flutterpi_tool #3: Flutter engine not found

eaceto, 2023-09-30:
Hello, how are you? First of all, great work! 

I am trying to migrate a project to this new build system, and I am getting an error when running:

```
flutterpi_tool build --arch=arm64 --release
Flutter engine binaries for engine a794cf2681c6c9fe7b260e0e84de96298dc9c18b are not available.
```

(arm64 for Raspberry Pi Zero 2 W with 64bit OS)

* Development Machine: Parallels ARM64 Ubuntu VM over macOS (M2)
* flutterpi_tool version: 0.1.2

* Flutter Doctor: 

` flutter doctor -v
[✓] Flutter (Channel stable, 3.13.6, on Ubuntu 22.04.3 LTS 5.15.0-84-generic, locale C.UTF-8)
    • Flutter version 3.13.6 on channel stable at /home/parallels/snap/flutter/common/flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision ead455963c (4 days ago), 2023-09-26 18:28:17 -0700
    • Engine revision a794cf2681
    • Dart version 3.1.3
    • DevTools version 2.25.0

[✗] Android toolchain - develop for Android devices
    ✗ Unable to locate Android SDK.
      Install Android Studio from: https://developer.android.com/studio/index.html
      On first launch it will assist you in installing the Android SDK components.
      (or visit https://flutter.dev/docs/get-started/install/linux#android-setup for detailed instructions).
      If the Android SDK has been installed to a custom location, please use
      `flutter config --android-sdk` to update to that location.

[✗] Chrome - develop for the web (Cannot find Chrome executable at google-chrome)
    ! Cannot find Chrome. Try setting CHROME_EXECUTABLE to a Chrome executable.

[✓] Linux toolchain - develop for Linux desktop
    • clang version 10.0.0-4ubuntu1
    • cmake version 3.16.3
    • ninja version 1.10.0
    • pkg-config version 0.29.1

[!] Android Studio (not installed)
    • Android Studio not found; download from https://developer.android.com/studio/index.html
      (or visit https://flutter.dev/docs/get-started/install/linux#android-setup for detailed instructions).

[✓] IntelliJ IDEA Ultimate Edition (version 2023.2)
    • IntelliJ at /home/parallels/Applications/idea-IU
    • Flutter plugin version 75.1.4
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart

[✓] Connected device (1 available)
    • Linux (desktop) • linux • linux-arm64 • Ubuntu 22.04.3 LTS 5.15.0-84-generic

[✓] Network resources
    • All expected network resources are available.

! Doctor found issues in 3 categories.
`

chochstedler, 2023-10-01:
I am seeing the same issue.

eaceto, 2023-10-01:
Of course that "tag" is not available at https://storage.googleapis.com/flutter_infra_release/, but neither any current tag newer than March of this year.

I'd love to help with this issue but I don't know where to start.

ardera, 2023-10-01:
Hey, this is because the 3.13.6 CI pipeline failed in my flutter-ci repo. (Just some required service crashed on my self-hosted actions runner)

The binaries should be available once https://github.com/ardera/flutter-ci/actions/runs/6328876140 has run through 👍

closed 2023-10-02 by ardera