ardera/flutterpi_tool #4: Providing --dart-define-from-file as a parameter to set secret values in the flutter app

mondxp, 2023-10-05:
Hey, first of all thank you for your great work on flutter pi. I really like deploying some smart home apps on my pi!

There is one parameter that I am missing on the flutterpi_tool. I want to be able to set some env variables via "--dart-define-from-file=" which is available in the flutter build command. This way I dont have any passwords or other secrets in my code.

Could you add this parameter in your tool?

ardera, 2024-01-17:
My bad, I added this some time ago in 9496b187e4b3a3f5b95a6cbb18d82aa4a17d5b63 but forgot to post it here. If there's anything else missing that's essential just lmk!

closed 2024-01-17 by ardera

mondxp, 2024-01-25:
Thank you very much!