ardera/flutterpi_tool #7: Flutter engine binaries not available on apple silicon

beniskopf, 2023-12-24:
Hi, I want to express my appreciation for your excellent work on Flutter Pi. whenever i try to use flutterpi_tool in a new hello world flutter project i got an error message:

`flutterpi_tool build --release`
`"Flutter engine binaries with version 3f3e560236539b7e2702f5ac790b2a4691b32d49 and target gen-snapshot-macOS-X64-armv7-generic-release.tar.xz are not available."`

here is my flutter doctor output

```
`flutter doctor -v
[✓] Flutter (Channel stable, 3.16.5, on macOS 14.0 23A344 darwin-arm64, locale en-DE)
    • Flutter version 3.16.5 on channel stable at /Users/daniel/Development/flutter/flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 78666c8dc5 (5 days ago), 2023-12-19 16:14:14 -0800
    • Engine revision 3f3e560236
    • Dart version 3.2.3
    • DevTools version 2.28.4

[✓] Android toolchain - develop for Android devices (Android SDK version 34.0.0)
    • Android SDK at /Users/daniel/Library/Android/sdk
    • Platform android-34, build-tools 34.0.0
    • Java binary at: /Applications/Android Studio.app/Contents/jre/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 11.0.13+0-b1751.21-8125866)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 15.0.1)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Build 15A507
    • CocoaPods version 1.14.3

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 2021.3)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.13+0-b1751.21-8125866)

[✓] IntelliJ IDEA Community Edition (version 2023.1.4)
    • IntelliJ at /Applications/IntelliJ IDEA CE.app
    • Flutter plugin version 74.0.4
    • Dart plugin version 231.9161.14

[✓] Connected device (2 available)
    • macOS (desktop) • macos  • darwin-arm64   • macOS 14.0 23A344 darwin-arm64
    • Chrome (web)    • chrome • web-javascript • Google Chrome 120.0.6099.129

[✓] Network resources
    • All expected network resources are available.

• No issues found!"
```
any ideas what i'm doing wrong?

ardera, 2023-12-24:
Hey, thanks I appreciate it!

That's my mistake, https://github.com/ardera/flutter-ci/commit/e8c2944e0e41999f8e7969d05c6964229f8ee77f was broken, so 3.16.5 doesn't work while 3.16.4 worked. Pushed a fix now, it should work once https://github.com/ardera/flutter-ci/actions/runs/7315339779 runs through

beniskopf, 2023-12-25:
hi, thanks for your help. i got rid of the original error but got a new one, i'm not sure if this problem is related to the original one.

`flutterpi_tool build --arch=arm64 --cpu=pi4 --release`
gives me
 `Target aot_elf_release failed: ProcessException: Failed to find "/Users/daniel/Development/flutter/flutter/bin/cache/artifacts/engine/flutterpi-gen-snapshot-darwin-x64-pi4-64-release/gen_snapshot" in the search path.
  Command: /Users/daniel/Development/flutter/flutter/bin/cache/artifacts/engine/flutterpi-gen-snapshot-darwin-x64-pi4-64-release/gen_snapshot 
`
i checked the engine folder and got this:

android-arm-profile/
android-arm-release/
android-arm/
android-arm64-profile/
android-arm64-release/
android-arm64/
android-x64-profile/
android-x64-release/
android-x64/
android-x86-jit-release/
android-x86/
common/
darwin-x64-profile/
darwin-x64-release/
darwin-x64/
flutterpi-engine-aarch64-generic-debug/
flutterpi-engine-aarch64-generic-debug_unopt/
flutterpi-engine-aarch64-generic-profile/
flutterpi-engine-aarch64-generic-release/
flutterpi-engine-armv7-generic-debug/
flutterpi-engine-armv7-generic-debug_unopt/
flutterpi-engine-armv7-generic-profile/
flutterpi-engine-armv7-generic-release/
flutterpi-engine-dbgsyms-aarch64-generic-debug/
flutterpi-engine-dbgsyms-aarch64-generic-debug_unopt/
flutterpi-engine-dbgsyms-aarch64-generic-profile/
flutterpi-engine-dbgsyms-aarch64-generic-release/
flutterpi-engine-dbgsyms-armv7-generic-debug/
flutterpi-engine-dbgsyms-armv7-generic-debug_unopt/
flutterpi-engine-dbgsyms-armv7-generic-profile/
flutterpi-engine-dbgsyms-armv7-generic-release/
flutterpi-engine-dbgsyms-pi3-64-profile/
flutterpi-engine-dbgsyms-pi3-64-release/
flutterpi-engine-dbgsyms-pi3-profile/
flutterpi-engine-dbgsyms-pi3-release/
flutterpi-engine-dbgsyms-pi4-64-profile/
flutterpi-engine-dbgsyms-pi4-64-release/
flutterpi-engine-dbgsyms-pi4-profile/
flutterpi-engine-dbgsyms-pi4-release/
flutterpi-engine-dbgsyms-x64-generic-debug/
flutterpi-engine-dbgsyms-x64-generic-debug_unopt/
flutterpi-engine-dbgsyms-x64-generic-profile/
flutterpi-engine-dbgsyms-x64-generic-release/
flutterpi-engine-pi3-64-profile/
flutterpi-engine-pi3-64-release/
flutterpi-engine-pi3-profile/
flutterpi-engine-pi3-release/
flutterpi-engine-pi4-64-profile/
flutterpi-engine-pi4-64-release/
flutterpi-engine-pi4-profile/
flutterpi-engine-pi4-release/
flutterpi-engine-x64-generic-debug/
flutterpi-engine-x64-generic-debug_unopt/
flutterpi-engine-x64-generic-profile/
flutterpi-engine-x64-generic-release/
flutterpi-gen-snapshot-darwin-x64-aarch64-generic-profile/
flutterpi-gen-snapshot-darwin-x64-aarch64-generic-release/
flutterpi-gen-snapshot-darwin-x64-armv7-generic-profile/
flutterpi-gen-snapshot-darwin-x64-armv7-generic-release/
flutterpi-gen-snapshot-darwin-x64-x64-generic-profile/
flutterpi-gen-snapshot-darwin-x64-x64-generic-release/
flutterpi-universal/
ios-profile/
ios-release/
ios/

what path could i use for arm64, pi4 settings? 
Merry Christmas! Warm regards, Daniel

ardera, 2023-12-25:
Thanks for reporting, there was a bug with selecting the correct gen_snapshot file. The gen_snapshots are only built for the generic targets (aarch64, armv7, x64), but flutterpi_tool still attempted to use a tuned gen_snapshot (for pi4-64), which wasn't there of course. Just fixed it and pushed, can you try again?

Merry christmas to you too!

beniskopf, 2023-12-26:
works like a charm, thanks

closed 2023-12-26 by beniskopf