ardera/flutterpi_tool #9: Flutter engine binaries for engine 4a585b79294e830fa89c24924d58a27cc8fbf406 are not available

SmartIotMaker, 2024-01-12:
I follow the example command to build app，but gived engine are not available：
1. git clone https://github.com/flutter/gallery.git flutter_gallery
2. cd flutter_gallery
3. git checkout d77920b4ced4a105ad35659fbe3958800d418fb9
4. flutter pub get
5. flutterpi_tool build --arch=arm64 --cpu=pi4 --release
when run step 5, there has the error return:
Flutter engine binaries for engine 4a585b79294e830fa89c24924d58a27cc8fbf406 are not available.

note: the flutterpi_tool is used "dart pub global activate flutterpi_tool" commond to install.

ardera, 2024-01-12:
The engine binaries for 3.16.6 are not uploaded yet, I'm currently making some changes to the CI runners so CI for that didn't run through yet. It should be up in maybe 2 hours, once https://github.com/ardera/flutter-ci/actions/runs/7502269865 runs through. Or you can use 3.16.5 for now

SmartIotMaker, 2024-01-13:
yes, now build work fine, thank you

follow this the build log:

$ flutterpi_tool build --arch=arm --cpu=pi4 --release
Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure you trust this source!
Downloading engine-armv7-generic-release.tar.xz...                  7.0s
Downloading engine-pi4-release.tar.xz...                         1,911ms
Downloading gen-snapshot-Linux-X64-armv7-generic-release.tar.xz...      2,264ms
Downloading universal.tar.xz...                                  1,825ms
Building Flutter-Pi bundle...

ardera, 2024-01-14:
Nice!

closed 2024-01-14 by ardera