ardera/flutterpi_tool #10: flutterpi_tool installation problem

quanggdungg0609, 2024-01-14:
when i used "flutter pub global activate flutterpi_tool" i received a message like this:
`Note: file is pinned to version 7.0.0 by flutter_tools from the flutter SDK.
See https://dart.dev/go/sdk-version-pinning for details.

Note: process is pinned to version 5.0.2 by flutter_tools from the flutter SDK.
See https://dart.dev/go/sdk-version-pinning for details.

Because flutterpi_tool <0.2.0 depends on file ^6.1.4 and every version of
  flutter_tools from sdk depends on file 7.0.0, flutterpi_tool <0.2.0 is
  incompatible with flutter_tools from sdk.
And because every version of flutterpi_tool depends on flutter_tools from sdk
  and flutterpi_tool >=0.2.0 depends on process ^4.2.4, every version of
  flutterpi_tool requires process ^4.2.4.
Because every version of flutterpi_tool depends on flutter_tools from sdk
  which depends on process 5.0.2, every version of flutterpi_tool requires
  process 5.0.2.
Thus, flutterpi_tool is forbidden.
So, because pub global activate depends on flutterpi_tool any, version solving
  failed.
`

ardera, 2024-01-14:
Seems like you're on flutter beta? Or master? Flutter stable is still using file 6.1.4: https://github.com/flutter/flutter/blob/stable/packages/flutter_tools/pubspec.yaml#L18

flutterpi_tool is only supported for flutter stable

quanggdungg0609, 2024-01-14:
thank you for the help, i installed the flutterpi_tool but i have errors when following the README.md to build a exemple app. I have searched all topics in the Issues section but have not found an answer. Here is the error:
[image]

ardera, 2024-01-15:
I saw your message on discord and replied to you there 😄 just adding it here for completeness:

> Hey, currently linux arm64 is not supported as a host platform. Though, I see you're running parallels; you can try running on macos natively, macOS arm64 (apple silicon) is supported as a host platform

ardera, 2024-01-17:
Can you try running `flutter pub global activate flutterpi_tool` again, and then rerunning the build?

If the error remains, the complete commandline output would be great, either copy & paste the output, or redirect the output to a file and attach that, etc. Thanks!

IceIce1030, 2024-02-06:
> Owner

https://github.com/ardera/flutterpi_tool/issues/10#issuecomment-1896421011

IceIce1030, 2024-02-06:
I also encountered the same situation, how should I deal with it?

ardera, 2024-06-04:
@IceIce1030 if you have a similar problem, please also post the complete output.

ardera, 2024-06-19:
@werner-scholtz fixed in af8b5f98ee6c23936b2907bf739f6fa1a738ba14 and 780bd021fb252cfbeb2c1b68a40416b5cdf53f6a

Can you try removing `bin/cache/flutter-pi.stamp` and `bin/cache/artifacts/flutter-pi` and try again?

werner-scholtz, 2024-06-28:
Hi, seems to be working now with the latest release (0.5.0).

closed 2025-09-18 by ardera