ardera/flutterpi_tool #11: Flutter 3.19.0 could not activate tool

AlphaBlend1975, 2024-02-17:
I try ti upgrade flutter to 3.19.0 
when activate flutterpi_tool got error

/xxxx/rdo$ flutter pub global activate flutterpi_tool
Note: file is pinned to version 7.0.0 by flutter_tools from the flutter SDK.
See https://dart.dev/go/sdk-version-pinning for details.

Note: process is pinned to version 5.0.2 by flutter_tools from the flutter SDK.
See https://dart.dev/go/sdk-version-pinning for details.

Because flutterpi_tool <0.2.0 depends on file ^6.1.4 and every version of flutter_tools from sdk depends on file
  7.0.0, flutterpi_tool <0.2.0 is incompatible with flutter_tools from sdk.
And because every version of flutterpi_tool depends on flutter_tools from sdk and flutterpi_tool >=0.2.0 depends on
  process ^4.2.4, every version of flutterpi_tool requires process ^4.2.4.
Because every version of flutterpi_tool depends on flutter_tools from sdk which depends on process 5.0.2, every
  version of flutterpi_tool requires process 5.0.2.
Thus, flutterpi_tool is forbidden.
So, because pub global activate depends on flutterpi_tool any, version solving failed.

ardera, 2024-02-19:
I had to update flutterpi_tool to be compatible with flutter 3.19.

You can update it by running `flutter pub global activate flutterpi_tool` again

payam-zahedi, 2024-02-20:
There is an analyzer issue in the v0.3.0

[image]

ardera, 2024-02-20:
is it still there for you? just checked and it's not there for me anymore

payam-zahedi, 2024-02-20:
Its fixed now 

[image]

payam-zahedi, 2024-05-03:
You can close this @ardera

closed 2024-05-03 by ardera