ardera/flutterpi_tool #12: Issue with building --profile & --release

payam-zahedi, 2024-02-20:
I'm not able to use the build command on the --release  and --profile mode.

The debug mode is ok I can build for it and run it.

Error and Stacktrace:

```
flutterpi_tool build --arch=arm64 --cpu=pi4 --release 
Target aot_elf_release failed: ProcessException: Failed to find
"/home/pi/sdk/flutter/bin/cache/artifacts/engine/flutterpi-gen-snapshot-linux-arm64-aarch64-generic-
release/gen_snapshot" in the search path.
  Command:
  /home/pi/sdk/flutter/bin/cache/artifacts/engine/flutterpi-gen-snapshot-linux-arm64-aarch64-generic
  -release/gen_snapshot
#0      LocalProcessManager.start (package:process/src/interface/local_process_manager.dart:56:7)
#1      ErrorHandlingProcessManager.start.<anonymous closure> (package:flutter_tools/src/base/error_handling_io.dart:686:24)
#2      _run (package:flutter_tools/src/base/error_handling_io.dart:556:20)
#3      ErrorHandlingProcessManager.start (package:flutter_tools/src/base/error_handling_io.dart:685:12)
#4      _DefaultProcessUtils.start (package:flutter_tools/src/base/process.dart:476:28)
#5      _DefaultProcessUtils.stream (package:flutter_tools/src/base/process.dart:496:35)
#6      GenSnapshot.run (package:flutter_tools/src/base/build.dart:75:26)
#7      AOTSnapshotter.build (package:flutter_tools/src/base/build.dart:222:56)
#8      AotElfBase.build (package:flutter_tools/src/build_system/targets/common.dart:322:52)
#9      _BuildInstance._invokeInternal (package:flutter_tools/src/build_system/build_system.dart:861:27)
<asynchronous suspension>
#10     Future.wait.<anonymous closure> (dart:async/future.dart:518:21)
<asynchronous suspension>
#11     _BuildInstance.invokeTarget (package:flutter_tools/src/build_system/build_system.dart:799:32)
<asynchronous suspension>
#12     Future.wait.<anonymous closure> (dart:async/future.dart:518:21)
<asynchronous suspension>
#13     _BuildInstance.invokeTarget (package:flutter_tools/src/build_system/build_system.dart:799:32)
<asynchronous suspension>
#14     FlutterBuildSystem.build (package:flutter_tools/src/build_system/build_system.dart:628:16)
<asynchronous suspension>
#15     buildFlutterpiBundle (package:flutterpi_tool/src/build_bundle.dart:375:20)
<asynchronous suspension>
#16     BuildCommand.run.<anonymous closure> (package:flutterpi_tool/src/build_bundle.dart:668:11)
<asynchronous suspension>
#17     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:153:19)
<asynchronous suspension>
#18     BuildCommand.run (package:flutterpi_tool/src/build_bundle.dart:632:5)
<asynchronous suspension>
#19     CommandRunner.runCommand (package:args/command_runner.dart:212:13)
<asynchronous suspension>
#20     main (package:flutterpi_tool/src/build_bundle.dart:777:5)
<asynchronous suspension>
#21     main (file:///home/pi/.pub-cache/hosted/pub.dev/flutterpi_tool-0.3.0/bin/flutterpi_tool.dart:4:3)
<asynchronous suspension>

Building Flutter-Pi bundle...                                           
Failed to build bundle.
```

payam-zahedi, 2024-02-20:
I'm using Raspberry Pi 4b with the following information:

```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

payam-zahedi, 2024-02-20:
@ardera do you have any Idea how can I fix this?

ardera, 2024-02-20:
Yeah, currently the tool is not supported on linux arm64. Github Actions doesn't provide ARM64 runners so it's a bit hard for me build the necessary binaries. But there should really be a nicer error message, this has come up multiple times in the past: #8 https://github.com/ardera/flutter-pi/issues/378

payam-zahedi, 2024-02-20:
How can I connect flutter pi to devtools is it even possible?

how can I check the performance ?

payam-zahedi, 2024-02-20:
I could build the bundles with My Mac. so you can close this issue

closed 2024-05-03 by payam-zahedi