ardera/flutterpi_tool #14: flutterpi_tool and macOS - valid environment?

pezi, 2024-03-12:
I am not sure - the docu of flutter-pi says this a linux only tool. But some issues of flutter-pi has a macOS context.

Therefore I tried flutterpi_tool also on macOS. Compiling release woks fine. But after the sync 

`./build/flutter_assets/kernel_blob.bin`

was missing. But this was a minor issue, taking this file from a linux build solves the problem.

ardera, 2024-03-12:
I'm running flutterpi_tool on MacOS myself and it works fine, the kernel_blob.bin file is only necessary for debug mode, so on other runtime modes it's okay if it's not there. What runtime mode did you use?

pezi, 2024-03-12:
Ups - wrong usage on my side.

closed 2024-03-12 by pezi

ardera, 2024-03-12:
All good 😄