ardera/flutterpi_tool #15: Error when flutter  pub global activate flutterpi_tool

xiaoxiaowesley, 2024-03-16:
When run `flutter  pub global activate flutterpi_tool`. 

Erro happened below:

```
Failed to build flutterpi_tool:flutterpi_tool:
../../.pub-cache/hosted/pub.flutter-io.cn/flutterpi_tool-0.3.0/lib/src/cache.dart:297:20: Error: The type 'HostPlatform' is not exhaustively matched by the switch cases since it doesn't match 'HostPlatform.windows_arm64'.
 - 'HostPlatform' is from 'package:flutter_tools/src/base/os.dart' ('flutter/packages/flutter_tools/lib/src/base/os.dart').
Try adding a wildcard pattern or cases that match 'HostPlatform.windows_arm64'.
    return switch (host) {

```

and my `flutter doctor` is :

```
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel master, 3.21.0-9.0.pre.26, on macOS 14.2.1 23C71 darwin-x64, locale
    zh-Hans-CN)
[!] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
    ✗ cmdline-tools component is missing
      Run `path/to/sdkmanager --install "cmdline-tools;latest"`
      See https://developer.android.com/studio/command-line for more details.
    ✗ Android license status unknown.
      Run `flutter doctor --android-licenses` to accept the SDK licenses.
      See https://flutter.dev/docs/get-started/install/macos#android-setup for more details.
[!] Xcode - develop for iOS and macOS (Xcode 15.0)
    ! CocoaPods 1.8.4 out of date (1.13.0 is recommended).
        CocoaPods is used to retrieve the iOS and macOS platform side's plugin code that
        responds to your plugin usage on the Dart side.
        Without CocoaPods, plugins will not work on iOS or macOS.
        For more info, see https://flutter.dev/platform-plugins
      To upgrade see https://guides.cocoapods.org/using/getting-started.html#updating-cocoapods
      for instructions.
[✓] Chrome - develop for the web
[✓] Android Studio (version 2022.3)
[✓] VS Code (version 1.87.2)
[!] Proxy Configuration
    ! NO_PROXY is not set
[✓] Connected device (2 available)
    ! Error: Browsing on the local area network for iPhone 12 pro . Ensure the device is
      unlocked and attached with a cable or associated with the same local area network as this
      Mac.
      The device must be opted into Developer Mode to connect wirelessly. (code -27)
[✓] Network resources

! Doctor found issues in 3 categories.
```

el-leonya, 2024-06-03:
I had the same problem on Linux
```
$ flutter pub global activate flutterpi_tool
[...]
Building package executables... (20.1s)
Failed to build flutterpi_tool:flutterpi_tool:
.asdf/installs/flutter/3.19.3-stable/.pub-cache/hosted/pub.dev/flutterpi_tool-0.4.0/lib/src/cache.dart:303:20: Error: Member not found: 'windows_arm64'.
      HostPlatform.windows_arm64 => 'Windows-ARM64',
                   ^^^^^^^^^^^^^
.asdf/installs/flutter/3.19.3-stable/.pub-cache/hosted/pub.dev/flutterpi_tool-0.4.0/lib/src/cache.dart:858:16: Error: The method 'getDirectorySize' isn't defined for the class 'OperatingSystemUtils'.
 - 'OperatingSystemUtils' is from 'package:flutter_tools/src/base/os.dart' ('.asdf/installs/flutter/3.19.3-stable/packages/flutter_tools/lib/src/base/os.dart').
Try correcting the name to the name of an existing method, or defining a method named 'getDirectorySize'.
    return _os.getDirectorySize(directory);
               ^^^^^^^^^^^^^^^^
```
Doctor:
```
$ flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 3.19.3, on Ubuntu 22.04.4 LTS 6.5.0-35-generic, locale en_US.UTF-8)
[✗] Android toolchain - develop for Android devices
    ✗ Unable to locate Android SDK.
      Install Android Studio from: https://developer.android.com/studio/index.html
      On first launch it will assist you in installing the Android SDK components.
      (or visit https://flutter.dev/docs/get-started/install/linux#android-setup for detailed instructions).
      If the Android SDK has been installed to a custom location, please use
      `flutter config --android-sdk` to update to that location.

[✓] Chrome - develop for the web
[✓] Linux toolchain - develop for Linux desktop
[!] Android Studio (not installed)
[✓] VS Code (version 1.89.1)
[✓] Connected device (2 available)
[✓] Network resources

! Doctor found issues in 2 categories.
```

el-leonya, 2024-06-04:
@ardera Thanks a lot!