ardera/flutterpi_tool #18: flutterpi_tool not building on Ubuntu 22.04.4 x86-64

chuck-h, 2024-05-19:
Build error on Lenovo Thinkpad T16 (AMD Ryzen 5 pro 6650u):
```
chuck@Lenny:~/Projects/flut$ dart pub global activate flutterpi_tool
+ _fe_analyzer_shared 65.0.0 (68.0.0 available)
+ analyzer 6.3.0 (6.5.0 available)
+ archive 3.3.2 (3.5.1 available)
+ args 2.4.2 (2.5.0 available)
+ async 2.11.0
+ boolean_selector 2.1.1
[...]
+ webkit_inspection_protocol 1.2.1
+ xml 6.5.0
+ yaml 3.1.2
+ yaml_edit 2.1.1 (2.2.1 available)
Building package executables... (10.0s)
Failed to build flutterpi_tool:flutterpi_tool:
../../.pub-cache/hosted/pub.dev/flutterpi_tool-0.4.0/lib/src/cache.dart:303:20: Error: Member not found: 'windows_arm64'.
      HostPlatform.windows_arm64 => 'Windows-ARM64',
                   ^^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dev/flutterpi_tool-0.4.0/lib/src/cache.dart:858:16: Error: The method 'getDirectorySize' isn't defined for the class 'OperatingSystemUtils'.
 - 'OperatingSystemUtils' is from 'package:flutter_tools/src/base/os.dart' ('../../Apps/flutter/packages/flutter_tools/lib/src/base/os.dart').
Try correcting the name to the name of an existing method, or defining a method named 'getDirectorySize'.
    return _os.getDirectorySize(directory);
               ^^^^^^^^^^^^^^^^
chuck@Lenny:~/Projects/flut$ 
```

kerberjg, 2024-05-20:
It's a duplicate of #16

ardera, 2024-05-20:
See my reply here: https://github.com/ardera/flutterpi_tool/issues/16#issuecomment-2121029748

In short, if you're trying use flutterpi_tool with flutter 3.19.x you need to add a version constraint the activate command: `dart pub global activate flutterpi_tool ^0.3.0`

ardera, 2024-05-20:
Can you tell me if this works for you? @chuck-h

chuck-h, 2024-05-20:
`flutter pub global activate flutterpi_tool ^0.3.0`  worked /\ @ardera

closed 2024-05-20 by chuck-h