ardera/flutterpi_tool #25: incorrect file permissions for flutter-pi, engine binaries when building bundle

ardera, 2024-07-04:
Sometimes the flutter-pi binary and flutter engine lack the executable permission bit in the built bundle.

tryy3, 2024-12-18:
I get this everytime when trying to run flutterpi_tool remotely from a windows machine to a RPi.
Guessing it's because we are copying files from the windows machine over to the RPi and therefor loses the permission bit.

As a workaround I just threw together a script that I can continously run on my RPi that looks for changes in the /tmp folder and add +x to flutter_pi.
Probably very hacky solution but it works for now.

If I continue to work on the RPi I might try to solve it in a better way but thought I would just share that this is a possibility.
I don't feel very comfortable sharing and recommending this workaround since it's probably quite hacky but thought I would share that I used a workaround in case anyone else is stuck on this.

From my testing, when using verbose debugging I was able to take the exact commands that is being ran and run it locally.
But when they are run over SSH then they fail without the +x permission bit on the file.

ardera, 2025-01-13:
@tryy3 I attempted to fix this some time ago with https://github.com/ardera/flutterpi_tool/commit/5453f53575d8fccd9bc067b818b1f12b13e04de6 (maybe you updated flutterpi_tool when trying the Pi 5?) but not sure that worked because I could never actually reproduce the problem myself 🤷‍♂️  I'll try on Windows.

tryy3, 2025-01-14:
Hmm maybe, I think I installed the flutterpi_tool after that commit.

I'll try to boot up the RPi Zero again and see if I still get the issue, I did a lot of changes so could be multiple factors to what made it work flawlessly now.