ardera/flutterpi_tool #42: Dependency Problem with Flutter 3.27.1

pezi, 2024-12-22:
```
flutter --version                         
Flutter 3.27.1 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 17025dd882 (6 days ago) • 2024-12-17 03:23:09 +0900
Engine • revision cb4b5fff73
Tools • Dart 3.6.0 • DevTools 2.40.2
```

```
flutter pub global activate flutterpi_tool
Note: file is pinned to version 7.0.0 by flutter_tools from the flutter SDK.
See https://dart.dev/go/sdk-version-pinning for details.

Note: http is pinned to version 1.2.2 by flutter_tools from the flutter SDK.
See https://dart.dev/go/sdk-version-pinning for details.

Because flutterpi_tool <0.2.0 depends on file ^6.1.4 and every version of
  flutter_tools from sdk depends on file 7.0.0, flutterpi_tool <0.2.0 is
  incompatible with flutter_tools from sdk.
And because every version of flutterpi_tool depends on flutter_tools from sdk
  and flutterpi_tool >=0.2.0 depends on http ^0.13.6, every version of
  flutterpi_tool requires http ^0.13.6.
Because every version of flutterpi_tool depends on flutter_tools from sdk
  which depends on http 1.2.2, every version of flutterpi_tool requires http
  1.2.2.
Thus, flutterpi_tool is forbidden.
So, because pub global activate depends on flutterpi_tool any, version solving
  failed.
Failed to update packages.
```

bdlukaa, 2024-12-27:
I am having the same issue with my [CI/CD](https://github.com/bluecherrydvr/unity/actions/runs/12511014653/job/34902535004).

DonatelloX, 2025-01-06:
For now, the only solution I have found is a downgrade:
```
flutter downgrade 3.22.0
flutter pub global activate flutterpi_tool
```

ardera, 2025-01-13:
Sorry for the wait. Just released version 0.6.0 which is compatible with 3.27: https://pub.dev/packages/flutterpi_tool

closed 2025-01-13 by ardera