ardera/flutterpi_tool #46: Dependency Conflict When Activating flutterpi_tool on Flutter 3.29.0

cocolocow, 2025-02-26:
Hello,

I'm encountering an issue when trying to activate flutterpi_tool using the command:

flutter pub global activate flutterpi_tool

On Flutter version 3.29.0, the activation fails with the following error:

```Because flutterpi_tool >=0.2.0 <0.6.0 depends on http ^0.13.6 and flutterpi_tool <0.2.0 depends on file ^6.1.4,
  flutterpi_tool <0.6.0 requires http ^0.13.6 or file ^6.1.4.
And because every version of flutter_tools from sdk depends on both file 7.0.1 and http 1.2.2, flutterpi_tool <0.6.0
  is incompatible with flutter_tools from sdk.
And because every version of flutterpi_tool depends on flutter_tools from sdk and flutterpi_tool >=0.5.2 depends on
  unified_analytics >=5.8.0 <7.0.0, every version of flutterpi_tool requires unified_analytics >=5.8.0 <7.0.0.
Because every version of flutterpi_tool depends on flutter_tools from sdk which depends on unified_analytics 7.0.1,
  every version of flutterpi_tool requires unified_analytics 7.0.1.
Thus, flutterpi_tool is forbidden.
So, because pub global activate depends on flutterpi_tool any, version solving failed.
Failed to update packages.``` 

Context and Analysis:
It appears that:

flutterpi_tool <0.6.0 requires either http ^0.13.6 or file ^6.1.4, while the flutter_tools from the Flutter SDK depends on http 1.2.2 and file 7.0.1.
There is also a conflict regarding unified_analytics: flutterpi_tool requires unified_analytics >=5.8.0 <7.0.0 whereas flutter_tools depends on unified_analytics 7.0.1.
These incompatibilities prevent the dependency resolution when trying to activate flutterpi_tool.

Steps to Reproduce:

Use Flutter version 3.29.0.
Run the command: flutter pub global activate flutterpi_tool
Observe the dependency resolution error shown above.

jasmin713, 2025-02-28:
For me, too. I get the same error.

DerKleinePunk, 2025-03-10:
Same Problem hier

Because flutterpi_tool >=0.2.0 <0.6.0 depends on http ^0.13.6 and flutterpi_tool <0.2.0 depends on file ^6.1.4,
  flutterpi_tool <0.6.0 requires http ^0.13.6 or file ^6.1.4.
And because every version of flutter_tools from sdk depends on both file 7.0.1 and http 1.2.2, flutterpi_tool <0.6.0
  is incompatible with flutter_tools from sdk.
And because every version of flutterpi_tool depends on flutter_tools from sdk and flutterpi_tool >=0.5.2 depends on
  unified_analytics >=5.8.0 <7.0.0, every version of flutterpi_tool requires unified_analytics >=5.8.0 <7.0.0.
Because every version of flutterpi_tool depends on flutter_tools from sdk which depends on unified_analytics 7.0.1,
  every version of flutterpi_tool requires unified_analytics 7.0.1.
Thus, flutterpi_tool is forbidden.
So, because pub global activate depends on flutterpi_tool any, version solving failed

karniv00l, 2025-03-12:
same issue

filiph, 2025-03-13:
Just ran into the issue today. Here's my attempt at a fix: https://github.com/ardera/flutterpi_tool/pull/48

PR #48 (2025-03-17): Fix dependency issue that prevented the tool to be installed after Flutter 3.29
One getter in `Artifacts` was renamed from `isLocalEngine` to `usesLocalArtifacts`. See here: https://github.com/flutter/flutter/commit/8b4d48715bddfdfbaa2bfcdd9950cd63d58d62db. The usage is the same.

Fixes #46
commit de70a658 (2025-03-17): Update a getter changed in flutter_tools

One getter in `Artifacts` was renamed from `isLocalEngine` to `usesLocalArtifacts`. See here: https://github.com/flutter/flutter/commit/8b4d48715bddfdfbaa2bfcdd9950cd63d58d62db. The usage is the same.

Fixes #46
closed 2025-03-17 by ardera