ardera/flutterpi_tool #60: Unable to test on RISCV Board

akifejaz, 2025-07-08:
I've [RISCV board](https://milkv.io/jupiter) and I'm testing flutterpi_tool to run app on this board. I'm able to setup this tool & create test app but when I run the app I get no display on my riscv machine & it throws the below errors.

I'had these steps:

```flutter pub global activate flutterpi_tool
flutterpi_tool devices add akif@machine1

flutter create test_app && cd test_app
flutterpi_tool run -d machine1
```

Which results
```
akif-10xe@akif10xe:~/projects/test_app$ flutterpi_tool run -d machine1
Launching lib/main.dart on machine1 (wireless) in debug mode...
Building Flutter-Pi bundle...                                            
Installing app on device...                                        71.1s
==============Locale==============
Flutter locale:
  default: en
  locales: en en.UTF-8 en_US en.UTF-8 en.UTF-8 en_US en.UTF-8
===================================
flutter-pi.c: Device "/dev/dri/card0" doesn't have a display connected. Skipping.
modesetting.c: Unknown DRM crtc property: color_matrix_coef
===================================
EGL information:
  version: 1.5
  vendor: Mesa Project
  client extensions: EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_a
ll_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_explicit_device EGL_EXT_platform_wayland 
EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_EXT_platform_xcb EGL_MESA_platform_gbm EGL_KHR_platform_gbm 
EGL_MESA_platform_surfaceless
  display extensions: EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_create_context_robustness EGL_EXT_image_dma_buf_impo
rt EGL_EXT_image_dma_buf_import_modifiers EGL_EXT_image_gl_colorspace EGL_EXT_query_reset_notification_strategy EGL_EXT_yuv_surface
 EGL_IMG_context_priority EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_fence_sync EGL_KHR_get_all_proc_a
ddresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_tex
ture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_partial_update EGL_KHR_r
eusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_imag
e EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display EGL_IMG_cl_image 
===================================
===================================
OpenGL ES information:
  version: "OpenGL ES 3.2 build 24.2@6603887"
  shading language version: "OpenGL ES GLSL ES 3.20 build 24.2@6603887"
  vendor: "Imagination Technologies"
  renderer: "PowerVR B-Series BXE-2-32"
  extensions: "GL_ANDROID_extension_pack_es31a GL_APPLE_texture_format_BGRA8888 GL_EXT_EGL_image_array GL_EXT_EGL_image_storage GL_
EXT_EGL_image_storage_compression GL_EXT_YUV_target GL_EXT_blend_func_extended GL_EXT_blend_minmax GL_EXT_buffer_storage GL_EXT_cle
ar_texture GL_EXT_clip_control GL_EXT_color_buffer_float GL_EXT_color_buffer_half_float GL_EXT_compressed_ETC1_RGB8_sub_texture GL_
EXT_conservative_depth GL_EXT_copy_image GL_EXT_discard_framebuffer GL_EXT_draw_buffers GL_EXT_draw_buffers_indexed GL_EXT_draw_ele
ments_base_vertex GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_EXT_gpu_shader5 GL_EXT_memory_object GL_E
XT_memory_object_fd GL_EXT_multi_draw_arrays GL_EXT_multisampled_render_to_texture2 GL_EXT_occlusion_query_boolean GL_EXT_polygon_o
ffset_clamp GL_EXT_primitive_bounding_box GL_EXT_read_format_bgra GL_EXT_robustness GL_EXT_sRGB GL_EXT_sRGB_write_control GL_EXT_se
parate_shader_objects GL_EXT_shader_framebuffer_fetch GL_EXT_shader_group_vote GL_EXT_shader_implicit_conversions GL_EXT_shader_io_
blocks GL_EXT_shader_non_constant_global_initializers GL_EXT_shader_pixel_local_storage GL_EXT_shader_pixel_local_storage2 GL_EXT_s
hader_texture_lod GL_EXT_shadow_samplers GL_EXT_sparse_texture GL_EXT_tessellation_point_size GL_EXT_tessellation_shader GL_EXT_tex
ture_border_clamp GL_EXT_texture_buffer GL_EXT_texture_cube_map_array GL_EXT_texture_format_BGRA8888 GL_EXT_texture_format_sRGB_ove
rride GL_EXT_texture_rg GL_EXT_texture_sRGB_R8 GL_EXT_texture_sRGB_RG8 GL_EXT_texture_sRGB_decode GL_EXT_texture_shadow_lod GL_EXT_
texture_storage_compression GL_EXT_texture_type_2_10_10_10_REV GL_EXT_unpack_subimage GL_IMG_framebuffer_downsample GL_IMG_program_
binary GL_IMG_read_format GL_IMG_shader_binary GL_IMG_texture_format_BGRA8888 GL_IMG_texture_npot GL_KHR_blend_equation_advanced GL
_KHR_blend_equation_advanced_coherent GL_KHR_debug GL_KHR_robustness GL_KHR_texture_compression_astc_ldr GL_NV_draw_instanced GL_NV
_instanced_arrays GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_EGL_image_external_essl3 GL_OES_EGL_sync GL_OES_compressed_ETC1
_RGB8_texture GL_OES_depth24 GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_draw_buffers_indexed GL_OES_draw_elements_ba
se_vertex GL_OES_element_index_uint GL_OES_fragment_precision_high GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_get_pro
gram_binary GL_OES_gpu_shader5 GL_OES_mapbuffer GL_OES_packed_depth_stencil GL_OES_required_internalformat GL_OES_rgb8_rgba8 GL_OES
_sample_shading GL_OES_sample_variables GL_OES_shader_image_atomic GL_OES_shader_io_blocks GL_OES_shader_multisample_interpolation 
GL_OES_standard_derivatives GL_OES_surfaceless_context GL_OES_tessellation_point_size GL_OES_tessellation_shader GL_OES_texture_3D 
GL_OES_texture_border_clamp GL_OES_texture_buffer GL_OES_texture_cube_map_array GL_OES_texture_float GL_OES_texture_half_float GL_O
ES_texture_half_float_linear GL_OES_texture_npot GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_vertex_
array_object GL_OES_vertex_half_float GL_OVR_multiview GL_OVR_multiview2 GL_OVR_multiview_multisampled_render_to_texture"
===================================
window.c: WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be t
he fitting device-pixel ratio for your display. 
Use the `-d` commandline parameter to specify the physical dimensions of your display.
display mode:
  resolution: 1680 x 1050
  refresh rate: 59.954250Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
  pixel format: (any)
pluginregistry.c: Initialized plugins: services, text input, raw keyboard plugin, gstreamer video_player, audioplayers, 
window.c: Flutter requested render surface before supplying surface dimensions.
gl_renderer.c: Choosing EGL config with pixel format ARGB 8:8:8:8...
Syncing files to device machine1 (wireless)...                     181ms

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

A Dart VM Service on machine1 (wireless) is available at: http://127.0.0.1:40633/r7uFb9gMq_U=/
The Flutter DevTools debugger and profiler on machine1 (wireless) is available at:
http://127.0.0.1:9100?uri=http://127.0.0.1:40633/r7uFb9gMq_U=/
egl_gbm_render_surface.c: using fourcc AR24 (ARGB 8:8:8:8) with modifier 0x0
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
```

Furthermore, When I checked the links I see this:
at: http://127.0.0.1:40633/r7uFb9gMq_U=/
[image]

at http://127.0.0.1:9100?uri=http://127.0.0.1:40633/r7uFb9gMq_U=/
[image]

The app seems to build properly and also running, but I see no display even though I've granted permissions for video/render using..
`sudo usermod -aG video,input,render $USER`

Can someone please help me regarding this?

akifejaz, 2025-07-09:
> It seems like GNOME is using the video output. You should be able to configure your system to boot into console (instead of desktop environment) using:

Thanks for reply, I resolved it by Switching to a Text Console (`Ctrl + Alt + F3`). After this I can see my test-app on the display attached to HDMI port of board. 

Just one problem, I'm unable to operate mouse here (as by default there is no-mouse input in text console), & to try out your solution I would need to reboot board (multiple users have this board access through ssh, I'll have to wait for them to finish so I can restart .. )

ardera, 2025-07-09:
There's also a way to achieve this without rebooting, I think it's `systemctl stop graphical.target` or `systemctl isolate multi-user.target` (though `isolate` will stop everything that doesn't explicitly belong to the multi-user target, so might be dangerous)

> Just one problem, I'm unable to operate mouse here (as by default there is no-mouse input in text console), & to try out your solution I would need to reboot board (multiple users have this board access through ssh, I'll have to wait for them to finish so I can restart .. )

Mouse input should work, if it doesn't, send me the commandline output or try: https://gist.github.com/ardera/e7ddf7c8e477d258a26dfa9c40002626

closed 2025-09-18 by ardera